Guenael/rtlsdr-wsprd #1: Adding NTP based frequency calibration

ha7dcd, 2016-06-13:
Might be a good idea to re-use the NTP frequency calibration from the RPi WSPR transmitter code:
https://github.com/JamesP6000/WsprryPi
Thus no need for TCXO or any sophisticated reference clock.

The guys in that repository used a very clever solution to make Hz precision calibration during the WSPR transmission using NTP_adjust fucntion. I've used too last time to calibrate my RTL-SDR receiver setup:
https://www.youtube.com/watch?v=SDCQ4_TwNWw

ha7dcd, 2016-06-13:
I was thinking about this and also a friend of mine noticed that as RPi and the RTL-SDR dongle uses different clock sources this is difficult. Thanksfully there is quite a big leakage from the RPi GPIO4 when WsperryPi transmit WSPR mode so it can be picked up with the RTL-SDR dongle for calibration. 

Just an idea.
Thanks for this great project!
73

Guenael, 2016-06-14:
Hi Zoltan,
Very good idea! But in the same time I don't want limit this project to Raspberry-Pi only (works fine on BeagleBone too). And for this idea, the code could be integrated or kept separately. Maybe a rtlsdr-pi-wsprd fork ?
In the same time, I think there is other way to calibrate RTL-SDR devices. In theory, you could check the timing of the buffer sent over USB, and deduce the frequency of the crystal (after one minute for example). But of course, it's work for the frequency alignment, but not for the frequency stability (on short periods / fuzzy drift).
I have no time to work on this, because of my TinyBeacon Rev.D version, but the idea is cool.
Thanks for your feedback.

ha7dcd, 2016-06-14:
Hello Guenael,

Checking the timing of the buffer sent over USB is a good idea, however I
believe the RTL-SDR's xtal stability is the problem, not the card computers
xtal this case.

An other cross platform and universal solution could be to calibrate to the
WWV reference stations: https://en.wikipedia.org/wiki/WWV_(radio_station)

closed 2016-06-24 by Guenael