Guenael/rtlsdr-wsprd #2: Support for up converters

bammi, 2016-06-13:
Please will you add support for up converters (HamitUp, Spyverter) as most people are more likely to use WSPR on the HF frequencies rather than VHF in you sample. It is very confusing what to enter on the command line, say i am wanting to monitor 20M WSPR frequency of 14.0956 MHz and my up converter is HamItUp that adds 125MHz. What would be the exact command line that i would issue for wsprd for this example??

Thanks again for your efforts.

Guenael, 2016-06-13:
Hi Jwahar,
Done. I added -u option, to specify your your upconverter frequency (ex. 125M). It does the same job of the offset option... but it's now a separate option.
./rtlsdr_wsprd -f 14.0956M -c A1XYZ -l AB12cd -g 29 -u 125M
Good day

closed 2016-06-13 by Guenael

bammi, 2016-06-13:
Thanks so much appreciate it!

Enjoy the DYACs - Sent from my IPhone