Guenael/rtlsdr-wsprd #4: "Caught signal 11" - looping after one period

sm3ulc, 2016-06-15:
I run "rtlsdr_wsprd -c xxx -l xxxx -g 20 -p -23 -o 125000000 -f 10138.6k" on my rpi-1B and it starts ok and run one period and right after just prints "Caught signal 11" in a loop. Unhandled crash? 

Trace of one run can be found here: (started some 77s before interval)

http://www.update.uu.se/~davidl/tmp/wsprd-trace

Guenael, 2016-06-16:
Well, it could be a buffer overflow or a problem with the driver callback (low or no response).
Your rpi 1 has only one core, and the decoding process takes a lot of CPU. I use 2 threads and if this  second thread used for decoding takes too much CPU, the driver could not be able to push the samples.
I'm not sure, but it could be good explanation.
Could you try with a rpi 2 ?
A solution could be to change the decoding thread priority (code side), but I have no guarantees.

Guenael, 2016-06-16:
BTW, thx for reporting this issue.

Guenael, 2016-06-16:
Benchmark : https://www.raspberrypistarterkits.com/2016/03/04/raspberry-pi-3-pi-2-pi-b-benchmark-review/

sm3ulc, 2016-06-16:
Have been running a few tests on the 1B:

110s raw 250 kS IQ and demodulated with csdr to wav with 12k samplerate. 
The decoding with wsprd (from wsjt) takes 4 seconds. 
rtlsd_wsprd takes about 23% of cpu when running.
Should be pretty fine with the cpu-power i think. Maybe adding lower priority to the decodingthread in order to not disturb the realtime sampling.

I don't own a pi2 so can't do testing on that model.

Btw.. demodulation with csdr as I run it via my script it takes 92 seconds just to demod on the 1B..

Guenael, 2016-06-16:
Ok, I will try to fix this using pthread_attr_setschedparam.
Note to myself : 
#include <pthread.h>
#include <sched.h>

pthread_attr_t tattr;
pthread_t tid;
int ret;
int newprio = 20;
sched_param param;

/* initialized with default attributes */
ret = pthread_attr_init (&tattr);

/* safe to get existing scheduling param */
ret = pthread_attr_getschedparam (&tattr, &param);

/* set the priority; others are unchanged */
param.sched_priority = newprio;

/* setting the new scheduling param */
ret = pthread_attr_setschedparam (&tattr, &param);

/* with new priority specified */
ret = pthread_create (&tid, &tattr, func, arg);

Guenael, 2016-06-25:
Hello,
Pthread priority added, but I have no RPi-1 to test this functionality.

Guenael, 2016-06-25:
Issue not solved, same problem reported (by VE2VAX)

ve2vax, 2016-06-25:
use  the  -S  (single pass mode, no subtraction (same as original wsprd)).

Solved the problem for me. 
 ./rtlsdr_wsprd -f 144.489M -c VE2VAX -l FN35JO -a 1 -S

ve2vax, 2016-06-25:
The problem , happen when dec_options.subtraction is set  =1

sm3ulc, 2016-06-25:
Is the code pushed?

Skickat från min Samsung-enhet

Guenael, 2016-06-25:
Yep, the code is updated, but I'm not sure if the priority works really. I'll need to take time to check this point carefully.
@ve2vax : The fano alog take more processing (option substraction). The program crash when the callback is not available. The only way to fix this is using a very low priority. But for now, it don't works...
BTW, thanks for helping.

sm3ulc, 2016-06-25:
I can test code monday night. Utc+1 :)
// David
Skickat från min Samsung-enhet

sm3ulc, 2016-07-01:
Hi all!

Finally had the ability to do a quick test on an rpi1b. Seem to get some
runs with no signal11 stops and some runs it seems ok.

73

// David

Guenael, 2016-07-03:
Hello,
I fixed a malloc bug present only with gcc5 (usually the case with Raspberrian).
Please retest. Thx.

Guenael, 2016-07-14:
I tested this update on RPi 1 during 48h without issue. But some versions of Raspberian react badly with USB, I added a note to the README file : 

_I noticed some disconnection problems with USB port while using Raspbian Jessie (2016-05-27-raspbian-jessie-lite.img). I rollbacked to Raspbian Wheezy (2015-02-16-raspbian-wheezy.img) and the problems solved by themself magically... For now, I have not investigated this issue, but if you experience some "Caught signal 11" error message, it could be this same problem. For now, I would recommend Raspbian Wheezy v2015-02-16._

Closing this issue. Thx

closed 2016-07-14 by Guenael

zodoczi, 2017-01-27:
Hello,

I see the same problem, using RPi 3 with 2017-01-11-raspbian-jessie.img

73,
Zoltan

[caught_signal_11.txt](https://github.com/Guenael/rtlsdr-wsprd/files/735529/caught_signal_11.txt)

rbw42, 2018-02-14:
Hi all,
I'm running a pi2 with jessie having the same problem.

Not sure if this code is still being maintained, but this problem appears to be a bug due to unsafe string handling in wsprsim_utils.c. There are several places where something could go wrong, but I have fixed this problem with a combination of stack smashing protection compiler options (which doesn't fix the actual problem) and changing the size of the temporary string message[23] in wsprsim_utils.c to be 24 chars, not 23. Also ensuring that string is properly terminated.

So around line 189, change the variable declarations and array initialisation to:
    char grid4[5], message[24];

    memset(message,0,sizeof(message));

I can send a patch if that is useful.

Regards,
Randall