Guenael/rtlsdr-wsprd #6: Issue with offset setting?

sm3ulc, 2016-09-18:
Have spent a number of hours of trying to calibrate center via the offset setting and are seing larger steps in the detected fq than the steps used.

I run a testsignal via WsprryPi which as a sideeffects gives of harmonies at 100 Hz intervals from the center signal calibrated to "exact" center of  the wspr-interval.
command: sudo ./wspr -p -40 SM3ULC JO89ul 10 30m 

I've patched the code for max swing detection +-100 => +-200 Hz. Also added a printout of the offset for troubleshooting. In the example i step offset from -100 100 in steps of 10. The result should be the detected signals "slides" downwards/decrease in fqs in steps of 10 (+- small error), right?

Loop used for calibration/troubleshooting:

for n in `seq -100 10 100`;do echo $n;./rtlsdr_wsprd -c SM0ULC -l JO98ul -g 20 -p 31 -o ${n} -u 125M -f 10138.7k | tee -a logg;sleep 0.5;done

Log:

160918-0544  -9.70  -0.93  10.140274  0   -100 SM3ULC JO89 10
160918-0544 -17.94  -1.00  10.140174  0   -100 SM3ULC JO89 10
160918-0548 -11.64  -0.89  10.140274  0   -90 SM3ULC JO89 10
160918-0548 -17.74  -0.91  10.140174  1   -90 SM3ULC JO89 10
160918-0552 -11.55  -0.98  10.140274  0   -80 SM3ULC JO89 10
160918-0552 -20.18  -1.02  10.140174  1   -80 SM3ULC JO89 10
160918-0556 -10.98  -0.98  10.140274  1   -70 SM3ULC JO89 10
160918-0556 -19.62  -0.91  10.140174  1   -70 SM3ULC JO89 10
160918-0600 -11.04  -0.91  10.140219  1   -60 SM3ULC JO89 10
160918-0604 -10.25  -0.93  10.140219  0   -50 SM3ULC JO89 10
160918-0612 -12.34  -1.06  10.140219  1   -30 SM3ULC JO89 10
160918-0612 -18.70  -1.02  10.140119  1   -30 SM3ULC JO89 10
160918-0616  -9.40  -0.91  10.140219  0   -20 SM3ULC JO89 10
160918-0616 -17.49  -0.93  10.140119  1   -20 SM3ULC JO89 10
160918-0620 -20.21  -0.98  10.140119  1   -10 SM3ULC JO89 10
160918-0624  -9.35  -0.95  10.140164  1     0 SM3ULC JO89 10
160918-0624 -17.73  -0.98  10.140264  1     0 SM3ULC JO89 10
160918-0628 -13.58  -0.89  10.140161  0    10 SM3ULC JO89 10
160918-0632  -8.48  -0.98  10.140160  1    20 SM3ULC JO89 10
160918-0632 -19.76  -0.98  10.140260  0    20 SM3ULC JO89 10
160918-0636 -10.13  -0.98  10.140160  1    30 SM3ULC JO89 10
160918-0636 -19.00  -0.89  10.140260  0    30 SM3ULC JO89 10
160918-0640  -9.26  -0.95  10.140159  0    40 SM3ULC JO89 10
160918-0640 -18.17  -0.98  10.140259  0    40 SM3ULC JO89 10
160918-0644 -16.43  -0.93  10.140104  0    50 SM3ULC JO89 10
160918-0644 -25.42  -0.95  10.140204  0    50 SM3ULC JO89 10
160918-0648 -10.55  -0.93  10.140104  0    60 SM3ULC JO89 10
160918-0648 -19.84  -0.98  10.140204  0    60 SM3ULC JO89 10
160918-0648 -27.67  -0.93  10.140304  0    60 SM3ULC JO89 10
160918-0652 -17.02  -0.91  10.140204  1    70 SM3ULC JO89 10
160918-0652 -19.38  -0.95  10.140304  0    70 SM3ULC JO89 10
160918-0656  -9.63  -0.93  10.140104  0    80 SM3ULC JO89 10
160918-0656 -18.50  -0.91  10.140204  0    80 SM3ULC JO89 10
160918-0656 -25.32  -0.93  10.140304  0    80 SM3ULC JO89 10
160918-0700  -6.97  -1.02  10.140104  0    90 SM3ULC JO89 10
160918-0700 -19.39  -1.02  10.140204  0    90 SM3ULC JO89 10
160918-0700 -26.04  -1.00  10.140304  0    90 SM3ULC JO89 10
160918-0704 -19.12  -0.95  10.140149  0   100 SM3ULC JO89 10
160918-0704 -23.61  -0.95  10.140248  0   100 SM3ULC JO89 10

Guenael, 2016-09-29:
Thx for this report,
Could you confirm the version used, it is up to date ? There was a float approx. error by the past. (git pull and check any diff).
Good day

sm3ulc, 2016-10-01:
cat .git/refs/heads/master
7aeedf3e1e25a324e64d3b3b9e15bb19372393d3
(cloned 16/9)

Is the version I'm running on the nodes. Also the swing extension (fmin/fmax) from the original code does not seem to work. Get about +-125 Hz, but that maybe never worked in the original wsprd-code. Checking diffs...

Guenael, 2016-10-01:
OK, I got it, I will check double -> float conversions. It could be an approx error on the main frequency. By the past, I updated FFTW to use float instead of double (only useful with a Raspberry 1, to save computation time). Thanks for your help.

Guenael, 2016-10-01:
I made additional tests with my similar project airspy-wsprd. The backend (fano.c  nhash.c  tab.c  wsprd.c  wsprd_utils.c  wsprsim_utils.c) is exactly the same, and there is no offset issue. So I suspect an alignment problem inside the rtlsdr library.

Guenael, 2016-10-01:
Confirmed, using GQRX, when the LO is adjusted, the frequency only change by gap. Example @144MHz : +189Hz, +62Hz, +183Hz, +62Hz. This is wired. I have to look the rtlsdr code...
Well, this is not related to this code and this project, but I will keep this issue open.

sm3ulc, 2016-10-02:
Great with confirmation at least. One featurerequest for calibration could be saving the IQ-block in the c2-format for possible viusalization. Then any signal could be used, not just WSPR.

Guenael, 2016-10-02:
I will add an extra digital offset alignment (outside the LO). About saving samples in the c2 format, the code is already here, but only used as debug purposes (commented). I will add an option, but using this option will crash on any raspberry pi, (problem related to the SD card access latency). 
Add. note : The offset problem is only on r820 devices. E4000 works fine for example.

Guenael, 2016-10-06:
Issue open for rtl-sdr : https://github.com/steve-m/librtlsdr/issues/33
About saving the sample, it crash the application on Raspberry Pi (high priority memory card access & USB rtl are not friends...)
If you want save some sample anyway, uncomment this code and rebuild : 

        /* DEBUG -- Save samples
        printf("Writing file\n");
        FILE* fd = NULL;
        fd = fopen("samples.bin", "wb");
        int r=fwrite(rx_state.iSamples, sizeof(float), samples_len, fd);
        printf("%d samples written file\n", r);
        fclose(fd);
        */

closed 2016-10-06 by Guenael