Guenael/rtlsdr-wsprd #10: "Caught signal 11" on RPi 3

zodoczi, 2017-01-28:
Hi,

I see the same problem, using RPi 3 with "2017-01-11-raspbian-jessie.img"

When using -S switch it disappears. 
Does anyone know what causing this?

Thanks,

`pi@raspberrypi:~/rtlsdr-wsprd $ ./rtlsdr_wsprd -f 14.0956M -c HA7DCD -l JN97ML -d 2
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Enabled direct sampling mode, input 2

Starting rtlsdr-wsprd (2017-01-27, 14:15z) -- Version 0.2
  Callsign     : HA7DCD
  Locator      : JN97ML
  Dial freq.   : 14095599 Hz
  Real freq.   : 14095599 Hz
  PPM factor   : 0
  Gain         : 2 dB
Wait for time sync (start in 38 sec)

Caught signal 11
Caught signal 11
... 1 MByte long text cut here ...
Caught signal 11
Caught signal 11`

Guenael, 2017-01-29:
Hello,
This bug is documented and not related to the code. Overall, the problem is related to the SDcard access. I will remove the multi-pass method to avoid this kind of problem. But for now, you can use "-S" option.
Good day

closed 2017-01-29 by Guenael

andrewbnortham, 2017-04-08:
Adding sudo in front of the command fixed the problem for me, raspberry pi 3. KE8FZT