Guenael/rtlsdr-wsprd #12: selecting device

zodoczi, 2017-06-12:
Is there a command line option to select a receiver device from multiple ones?
In "rtl_sdr" the "-d" command line switch works for that, but in rtlsdr-wsprd it looks like doesn't have effect.

73 and Thanks,
Zoltan

Guenael, 2017-06-21:
Hello Zoltan,
It's possible to add this feature, using the rtl_index.
The -d is already used for direct sampling. But I will see to include this feature the next week-end.
73

Guenael, 2017-06-23:
Done, it should work now using "-i" option. To use your second device, use "-i 1".
Exemple : 

$ ./rtlsdr_wsprd -f 144.489M -c VA2GKA -l FN35fm -g 29 -i 1
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 1: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Starting rtlsdr-wsprd (2017-06-23, 04:31z) -- Version 0.2
  Callsign     : VA2GKA
  Locator      : FN35fm
  Dial freq.   : 144489000 Hz
  Real freq.   : 144489000 Hz
  PPM factor   : 0
  Gain         : 29 dB
Wait for time sync (start in 28 sec)

closed 2017-06-23 by Guenael