Guenael/rtlsdr-wsprd #13: Adding date and time to the decoding result.

DickvanNoort, 2017-06-24:
Hi Guenael,
I see you are currently busy are making changes.
Could you consider adding the date and time to the results like this:

2017-06-24, 06:38z Spot : -21.80 -0.55  28.125172  0  G7FYO IO82 33
2017-06-24, 06:40z Spot : -15.65 -1.85  28.125190  0  IK1WVQ JN44 27
2017-06-24, 06:42z, No spot
2017-06-24, 06:44z Spot : -17.16 -0.91  28.125129  0  EA1FAQ IN71 37

It makes it much easier to tune your setup and interpret the results.

Also using this command 
./rtlsdr_wsprd -f 28.1236M -c PA2015 -l JO22XF -g 29 -p +4 -o 1000 -S >> "logs/wsprd_$(date +"%Y%m%d%H%M").log" 2>&1 &
with redirected the output results in a unique file with the settings and the result.
Very handy..

Cheers,
Dick (PA-2015)
the Netherlands

Guenael, 2017-07-01:
Hello,
It's easy to implement, but the logs are already uploaded on WsprNet with all these information.
Anyway, I push a mod soon.
Good day,

closed 2017-07-01 by Guenael

DickvanNoort, 2017-07-01:
Thank you, much appreciated.
Having all the details in a local file makes it easier to experiment with settings and equipment..

cheers,
Dick PA2015

From: Guenael, VA2GKA 
Sent: 1 July, 2017 04:03
To: Guenael/rtlsdr-wsprd 
Cc: DickvanNoort ; Author 
Subject: Re: [Guenael/rtlsdr-wsprd] Adding date and time to the decoding result. (#13)

Hello,
It's easy to implement, but the logs are already uploaded on WsprNet with all these information.
Anyway, I push a mod soon.
Good day,

IZ7BOJ, 2021-11-25:
Hi,
I forked a version which implements this feature: https://github.com/IZ7BOJ/rtlsdr-wsprd 
I needed timestamps tu run a web dashboard which shows received spots: https://github.com/IZ7BOJ/WSPRd_web_interface
I hope this helps
Alfredo