Guenael/rtlsdr-wsprd #14: Select device based on serialnumber or device-ID.

DickvanNoort, 2017-06-24:
For those having multiple RTLSDR dongles on their RPi, it is very handy if the intended dongle can be selected using its serial-number.

Programs that support his feature and are available in github are  
acarsdec at https://github.com/TLeconte/acarsdec 
and 
dumpvdl2 at https://github.com/szpajder/dumpvdl2 

Cheers,
Dick (PA-2015)

LanWolf, 2018-02-13:
An other additional request would be to use keenerd's rtl-sdr implementation, which support direct sampling on the RTL-SDR blog sticks. 
(https://github.com/keenerd/rtl-sdr)

Guenael, 2021-12-01:
-- Sorry for the (extra-long) delay, I just recovered my access to GitHub, and I was off radio activities --

Hi all,
I guess I missed this comment before adjusting some code by the past. This feature should be available with this option: 
```
"\t-i device index (in case of multiple receivers, default: 0)\n"
```

When I committed the first version of this tool, a Raspeberry Pi One was barely able to decode the spots in real-time (under 1min30). So, using different rtl was not the idea, but I guess it should work with this option now.

73

closed 2021-12-01 by Guenael