Guenael/rtlsdr-wsprd #15: Many entries like this " Spot : -19.11 2.33  28.125194  0  <...> A000AA 63"

DickvanNoort, 2017-08-27:
Hello Guenael,

My log shows quite a fewl entries like this
"Spot : -19.11 2.33  28.125194  0  <...> A000AA 63"
As a very beginner in C programming I searched the source and found this in "wsprd_utils.c"

      // I don't know what to do with these... They show up as "A000AA" grids.
      if( ntype == -64 ) noprint=1;

Are these lines referring to the entries I find in my the log?

73,
Dick

binaryer, 2018-03-31:
same !
Spot : -17.82 -4.60  50.294402  0  <...> A000AA 63  
Spot : -17.89 0.67  50.294402  0  <...> A000AA 63  
Spot : -16.81 1.24  50.294402  0  <...> A000AA 63  
Spot : -19.18 -3.34 144.490402  0  <...> A000AA 63  
Spot : -20.86 -3.28 432.301502  0  <...> A000AA 63

binaryer, 2018-03-31:
same !
```
Spot : -17.82 -4.60 50.294402 0 <...> A000AA 63
Spot : -17.89 0.67 50.294402 0 <...> A000AA 63
Spot : -16.81 1.24 50.294402 0 <...> A000AA 63
Spot : -19.18 -3.34 144.490402 0 <...> A000AA 63
Spot : -20.86 -3.28 432.301502 0 <...> A000AA 63
```

Swell1, 2020-12-24:
Analyzing the A000AA results, I am beginning to think that they are actually real spots. However either the script is not able to correctly decrypt certain combinations, OR some transmissions themselves have faulty encryption, OR the signal is too distorted for recovery. 
This is based on the way is which certain frequencies are repeated often, suggesting that particular wspr beacons are being heard repeatedly but not being decrypted

Swell1, 2020-12-24:
Another point is that, being also a beginner in C, and with reference to the first post above, the "ntype" for the "A000AA" spots is alw

Swell1, 2020-12-24:
Sorry
Is always 63 whereas Guenael's comment would suggest it should be 64?

Guenael, 2021-12-01:
-- Sorry for the (extra-long) delay, I just recovered my access to GitHub, and I was off radio activities --

Hi all,

Interesting issue, I never experienced this kind of bad report.
Can you provide a sample (wav file or c2 file) ?

Thanks, 73,

Guenael, 2021-12-05:
There are other reports with this `A000AA` pattern, and it could be a limitation issue.

If I understand correctly, when you transmit a callsign with a four character suffix, you can get this pattern.

closed 2021-12-05 by Guenael