Guenael/rtlsdr-wsprd #23: Running error on openwrt

d9394, 2019-10-18:
[image]
 I have successed compile rtlsdr_wsprd from source on the openwrt(version 18.06.02) and cpu hardware is mt7621 chipset,  when I run it appear a error

d9394, 2019-10-18:
[image]
after add some checkpoint

d9394, 2019-10-18:
[putty.log](https://github.com/Guenael/rtlsdr-wsprd/files/3744939/putty.log)
 log file with command: strace -tt rdlsdr_wsprd

Guenael, 2021-12-01:
-- Sorry for the (extra-long) delay, I just recovered my access to GitHub, and I was off radio activities --

Hi @d9394 ,

Well, the end of the log didn't help too much... 
```
00:49:55.977082 nanosleep({tv_sec=0, tv_nsec=250000000}, RX done! [Buffer size: 43500]
Search & decode the signal
 <unfinished ...>) = ?
00:49:56.035480 +++ killed by SIGSEGV +++
Segmentation fault
]0;root@OpenWrt: ~root@OpenWrt:~# 
```

Did you try with a similar project based on files only ?
Ex. https://github.com/k9an/old_wsprcan

It could be also an allocation issue. This MR (merged) could fix an issue related to this, but this one is pretty rare.
https://github.com/Guenael/rtlsdr-wsprd/pull/22

73

closed 2021-12-01 by Guenael