Guenael/rtlsdr-wsprd #27: OS version used with rtlsdr-wsprd

itstechpro, 2021-07-27:
I noticed that rtlsdr-wsprd was developed on Wheezy.  I'm running Buster on a Pi 3B+ and have also tried it on a Pi ZeroW.  Managed to get wsprd working but it usually stops after one to three hours.  If I restart without rebooting, wsprd sometimes fails to upload spots.

Swell1, 2021-07-28:
I have used a Pi zero W till 4 months back with the latest raspbian, and continuously over days.

During March I noticed an upload problem with WSPR [at a temporary QTH] but put it down to the reported heavy load on the server since an occassional upload did take place. During last year I had lots of uploads with the same setup.

Have not started WSPR again after return to my QTH, will do so soon.

So not sure of why you are facing this problem.

73
Sunil, VU3ZAN

  

&mdash;

itstechpro, 2021-07-28:
Thank you.  Would appreciate your observations when you get a chance to 
test.  I'm using VNC with a wifi connection to the Pi ZeroW.
73,
Jules-K2KGJ

IZ7BOJ, 2021-11-25:
Hi,
I managed to run rtlsdr_wsprd on Jessie distro and RPi 3B, but only with option "-S". A friend of mine managed to run it on Stretch distro. I'm trying to run on a Buster distro and RPi 3B but I can't. It doesn't decode anything, and the other strange behaviour is that I can't kill the application. If I do "sudo kill <pid>", it stays alive!
Anyone could help me?
73' Alfredo IZ7BOJ

Guenael, 2021-12-01:
-- Sorry for the (extra-long) delay, I just recovered my access to GitHub, and I was off radio activities --

Hi all,
About the stability issue, this MR below (just merged) could fix it. Try to rebuild the binary and test plz.
https://github.com/Guenael/rtlsdr-wsprd/pull/22

About the OS itself, I'm not an expert with RPi OS vs a regular Linux, it should didn't change anything, but I will test the latest version of Raspberrian this week-end.
73,

Guenael, 2021-12-03:
@IZ7BOJ Did you used another SD card? It could be an issue with the file system when using extensive search. Try again with -q -s options.
I compiled the latest version on a fresh `Raspbian GNU/Linux 10 (buster)` on a Raspberry Pi 4 and it works fine.

IZ7BOJ, 2021-12-03:
@Guenael  You anticipated me. I was writing that rtlsdr_wsprd works fine on a "bullseye" fresh image that I installed yesterday, sto the problem was on my previous operative system. I upgraded from stretch to buster with "apt dist-upgrade", maybe something went wrong.
Sorry for the false alarm and thank you for your effort

Guenael, 2021-12-03:
Happy to read that. Good job!

closed 2021-12-03 by Guenael

Guenael, 2021-12-03:
@IZ7BOJ I'm writing rtlsdr-ft8, do you want to test it?
(right now, I don't have any decent setup to use...)

IZ7BOJ, 2021-12-04:
@Guenael I'm not very familiar with ft-8, but I will try!

Guenael, 2021-12-04:
@IZ7BOJ Well, this digimode drained all the traffic from CW ops... If you like easy contact, this is the place :)

Guenael, 2021-12-15:
@IZ7BOJ what version did you used exactly?
The legacy one ?
https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2021-12-02/