Guenael/rtlsdr-wsprd #30: Setup a build pipeline for this project

Guenael, 2021-12-04:
TODO: format check + build pipeline, to check any MR on master + build release system

Some idea to start with GitHub (not very familiar, I only known GitLab CI/CD system)

```
name: CI Build pipeline
on: [push]
jobs:
  build:
    runs-on: ubuntu-latest
    steps:
      - run: echo "INFO -- This job was automatically triggered by a ${{ github.event_name }} event on a ${{ runner.os }} server"
      - run: echo "INFO -- Branch name: ${{ github.ref }} Repo name: ${{ github.repository }}."
      - name: Check out repository code
        uses: actions/checkout@v2
      - name: Install OS dependencies
        run: apt-get install build-essential cmake libfftw3-dev libusb-1.0-0-dev libcurl4-gnutls-dev librtlsdr-dev
      - name: Install cpplint # install cpplint with pip before
        run: pip install cpplint
      - name: Build
        run: make
      - name: Test
        run: echo "TODO"
      - name: DBG_status
        run: echo "INFO -- Status: ${{ job.status }}"
```

dforsi, 2021-12-04:
You may also consider the following tools based on Clang and available via apt-get:

- cppcheck
- iwyu
- clang-tools-11 (for scan-build)

All them seem to work fine with the current Makefile and I am about to open an issue with warnings found with cppcheck.

(BTW: glad to see that you are active again)

Guenael, 2021-12-04:
Thanks for your comment & suggestion!
I integrated cppcheck & cpplint in the build pipeline. I will take a look on iwyu

closed 2021-12-04 by Guenael