Guenael/rtlsdr-wsprd #41: multi-site upload

Guenael, 2021-12-04:
ex. https://pskreporter.info

Guenael, 2021-12-04:
Exemple with PSKreporter intergrated:
```
WSJT-X, fldigi, Digital Master 780, JT65-HF, ROS, JS8Call, OpenWebRX
```

https://github.com/w1hkj/fldigi/blob/master/src/spot/pskrep.cxx
https://pskreporter.info/pskdev.html
https://pskreporter.info/pskreporter.zip  // DLL, useless

Guenael, 2021-12-18:
@IZ7BOJ
Hi Alfredo,
Instead of parsing the console to collect the logs/spots, it could be possible to upload the spots directly on your web-site with an additional https request.
LMK

IZ7BOJ, 2021-12-18:
Thanks Guenael,
my web server runs on a raspberry and doesn't have a database.
The only way that I have to keep the spot history and show it on a web page is a log file.
Actually wsprd doesn't have a log, so my workaround is a stdout redirect to a file, but I had in mind to add a log option to rtlsdr_wsprd and make a pull request on github, as soon as I have time for it :-)
Bests,
Alfredo

Guenael, 2021-12-31:
I created a branch for PSKreporter testing:
https://github.com/Guenael/rtlsdr-wsprd/tree/pskreporter