Guenael/rtlsdr-wsprd #67: Feature request - run script

mkarliner, 2021-12-10:
I'd like to run something like https://github.com/JamesP6000/WsprryPi
in tandem with  rtlsdr-wsprd. 

To do this, I'd like an option to run a script at defined intervals, so I
can trigger a T/R relay and run WsprryPi in single shot mode.

This would make an absolutely tiny full WSPR station.

Guenael, 2021-12-11:
Think you can do this easily, with this option:
```
-n max iterations (default: 0 = infinite loop)
```

If you set this option to one, and if your script wait one rtlsdr-wsprd to exit, then you can switch & TX. It could work.

I noticed rtlsdr-wsprd is reluctant to quit properly... Another bug related to my management with pthread. Not an easy one.

closed 2021-12-11 by Guenael

sm3ulc, 2021-12-11:
Great!