Guenael/rtlsdr-wsprd #95: 0.4.0 no decodes

mkarliner, 2021-12-16:
./rtlsdr_wsprd -w 0.4test -d 2 -f 20m -c G8LKD -l IO91VN
Saving IQ file planned with prefix: 0.4test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Enabled direct sampling mode, input 2

Starting rtlsdr-wsprd (2021-12-16, 18:13z) -- Version 0.4.0
  Callsign     : G8LKD
  Locator      : IO91VN
  Dial freq.   : 14095600 Hz
  Real freq.   : 14095600 Hz
  PPM factor   : 0
  Gain         : 29 dB
Wait for time sync (start in 44 sec)

              Date  Time(z)    SNR     DT       Freq Dr    Call    Loc Pwr
No spot 2021-12-16 18:14z
No spot 2021-12-16 18:16z
No spot 2021-12-16 18:18z
No spot 2021-12-16 18:20z
No spot 2021-12-16 18:22z
No spot 2021-12-16 18:24z
[04test.zip](https://github.com/Guenael/rtlsdr-wsprd/files/7729632/04test.zip)

No spot 2021-12-16 18:26z

Guenael, 2021-12-17:
About this post, did you fixed the `-d 2` by yourself? Or is it related to the other issue with the direct sampling?

closed 2021-12-19 by Guenael

mkarliner, 2021-12-19:
I just hard coded a 2 to check the direct sampling. As I checked the code in ft8
and saw that it was correct there, I left it to you to do the.

Guenael, 2021-12-19:
@mkarliner I roll-backed this part, you should be able to use `2` directly in the options.
Change here and here:
https://github.com/Guenael/rtlsdr-ft8d/blob/main/rtlsdr_ft8d.c#L982
https://github.com/Guenael/rtlsdr-ft8d/blob/main/rtlsdr_ft8d.c#L1082