Guenael/rtlsdr-wsprd #131: wsprnet.org post results in 0 of one uploaded from rtlsdr-wsprd and when placed in safari browser

bradshawlupton, 2022-10-18:
http://wsprnet.org/post?function=wspr&rcall=wb2tev&rgrid=FN41&rqrg=28.126205&date=221018&time=0016&sig=-16&dt=-1.0&tqrg=28.126205&tcall=WB2TEV&tgrid=FN40&dbm=8&version=rtlsdr-054&mode=2

I compared this to the specification and it looks good, any suggestions will be appreciated.

Reception and decoding is perfect.

Bradshaw k1TE, Buzzards Bay, MA

Guenael, 2022-10-18:
@bradshawlupton Usually WSPRnet reject your post if it is not aligned within 5 minutes. Please double check your date/time when using wsprd, and if you do a manual post with `curl` or your web-browser, adjust the date. GL

Swell1, 2022-11-17:
Dear Friends
Sorry I&#39;m somehow not able to originate a mail to you using this system, so I am replying to all.

Subject: Bullseye on RPi needs addition to script

Hello
I&#39;ve been using your wonderful script for 2 years or more. I have used two RPi Zero W with Buster on them and the wsprd.
This week I resumed using them after a long time and updated and upgraded both. In one case the udev rules problem came. So I tried with Bullseye on that RPZW.

Error came :
1. Dev rules not being installed, install them with _DINSTALL_UDEV_RULES=ON
2. Building with usfbs zero-copy support disabled, use _DENABLE_ZEROCOPY=ON to enable
3 At the end the message was &quot;warning: manually specified variables were not used by the project: CMAKE_INSTALL_PREFIX

I deleted the rtl-sdr and rtlsdr-wsprd folders and rebooted. Fumbled a bit. Then finally I added the two items above in the middle of your script line that starts with &#39;cmake -DCMAKE_ &#39;
Thank God, everything worked perfectly.
Screenshots attached for your reference. I hope this will help to make any necessary changes to instructions on GitHub, regarding use of Bullseye for WSPRD

Regards
Sunil Aruldas
VU3ZAN

  

&mdash;

Guenael, 2024-10-12:
Extra `-DINSTALL_UDEV_RULES=ON -DENABLE_ZEROCOPY=ON` added on the documenation.
Thanks @Swell1 / VU3ZAN

But long story short, RTLSDR lib/apps are supposed to work when you install rtlsdr-wsprd. I added some instructions in the documentation, but I guess I shot myself in the foot...

closed 2024-10-12 by Guenael