Guenael/rtlsdr-wsprd #137: Clang error when compiling

morphyrichards, 2024-07-11:
Trying to compile rtlsdr-wsprd on a Raspberry Pi 3 running OS ver 6.6.20+rpt-rpi-v8, following the instructions in the readme.

I've tried deleting both rtl_sdr and recompiling this and then wsprd but always get this error below.

```
pi@rtlSdrTcpPi:~/rtlsdr-wsprd $ make
clang -O3 -std=gnu17 -c rtlsdr_wsprd.c -o rtlsdr_wsprd.o
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: clang -O3 -std=gnu17 -c rtlsdr_wsprd.c -o rtlsdr_wsprd.o
1.      <eof> parser at end of file
2.      Code generation
3.      Running pass 'Function Pass Manager' on module 'rtlsdr_wsprd.c'.
4.      Running pass 'AArch64 Assembly Printer' on function '@postSpots'
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm3sys15PrintStackTraceERNS_11raw_ostreamEi+0x44)[0x7fa0cfe4dc]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm3sys17RunSignalHandlersEv+0x70)[0x7fa0cfc4b4]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(+0xdcb1c0)[0x7fa0c2b1c0]
linux-vdso.so.1(__kernel_rt_sigreturn+0x0)[0x7fa95777b0]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(+0x132a9f0)[0x7fa118a9f0]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm11raw_ostream5writeEPKcm+0x164)[0x7fa0cdb49c]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm9MCContext16createTempSymbolERKNS_5TwineEb+0xb4)[0x7fa2091048]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm9MCContext16createTempSymbolEv+0x3c)[0x7fa20913f0]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm16MCObjectStreamer18emitCFIEndProcImplERNS_16MCDwarfFrameInfoE+0x1c)[0x7fa20b5c6c]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(+0x165f4c0)[0x7fa14bf4c0]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm10AsmPrinter16emitFunctionBodyEv+0x264c)[0x7fa14a0b6c]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(+0x2764ba4)[0x7fa25c4ba4]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm19MachineFunctionPass13runOnFunctionERNS_8FunctionE+0x140)[0x7fa105122c]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm13FPPassManager13runOnFunctionERNS_8FunctionE+0x26c)[0x7fa0e34390]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm13FPPassManager11runOnModuleERNS_6ModuleE+0x3c)[0x7fa0e3af70]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm6legacy15PassManagerImpl3runERNS_6ModuleE+0x7b4)[0x7fa0e34d98]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang17EmitBackendOutputERNS_17DiagnosticsEngineERKNS_19HeaderSearchOptionsERKNS_14CodeGenOptionsERKNS_13TargetOptionsERKNS_11LangOptionsEN4llvm9StringRefEPNSE_6ModuleENS_13BackendActionESt10unique_ptrINSE_17raw_pwrite_streamESt14default_deleteISK_EE+0xad0)[0x7fa793c680]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(+0x1b06f20)[0x7fa7bf6f20]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang8ParseASTERNS_4SemaEbb+0x210)[0x7fa6b36f1c]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang14FrontendAction7ExecuteEv+0x80)[0x7fa84f4444]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang16CompilerInstance13ExecuteActionERNS_14FrontendActionE+0x328)[0x7fa8467d28]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang25ExecuteCompilerInvocationEPNS_16CompilerInstanceE+0x218)[0x7fa8567d20]
clang(_Z8cc1_mainN4llvm8ArrayRefIPKcEES2_Pv+0x810)[0x413608]
clang[0x411df4]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(+0x207c5fc)[0x7fa816c5fc]
/lib/aarch64-linux-gnu/libLLVM-14.so.1(_ZN4llvm20CrashRecoveryContext9RunSafelyENS_12function_refIFvvEEE+0xcc)[0x7fa0c2ae94]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZNK5clang6driver10CC1Command7ExecuteEN4llvm8ArrayRefINS2_8OptionalINS2_9StringRefEEEEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPb+0x120)[0x7fa816c000]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZNK5clang6driver11Compilation14ExecuteCommandERKNS0_7CommandERPS3_+0x2b0)[0x7fa8140f50]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZNK5clang6driver11Compilation11ExecuteJobsERKNS0_7JobListERN4llvm15SmallVectorImplISt4pairIiPKNS0_7CommandEEEE+0x7c)[0x7fa814118c]
/lib/aarch64-linux-gnu/libclang-cpp.so.14(_ZN5clang6driver6Driver18ExecuteCompilationERNS0_11CompilationERN4llvm15SmallVectorImplISt4pairIiPKNS0_7CommandEEEE+0xd8)[0x7fa81550c4]
clang(main+0x23d4)[0x411648]
/lib/aarch64-linux-gnu/libc.so.6(+0x27780)[0x7f9f9d7780]
/lib/aarch64-linux-gnu/libc.so.6(__libc_start_main+0x98)[0x7f9f9d7858]
clang(_start+0x30)[0x40eef0]
clang: error: clang frontend command failed with exit code 139 (use -v to see invocation)
Debian clang version 14.0.6
Target: aarch64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
clang: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: /tmp/rtlsdr_wsprd-457e6f.c
clang: note: diagnostic msg: /tmp/rtlsdr_wsprd-457e6f.sh
clang: note: diagnostic msg:

********************
make: *** [Makefile:18: rtlsdr_wsprd.o] Error 139
```

joevk5ei, 2024-07-30:
Im getting the same error, any progress on this issue would be great please.

morphyrichards, 2024-08-03:
Just successfully compiled it on an old rpi-2 running Bullseye, but that old pi2 is only 32 bit so uses the 32 bit kernel.

Guenael, 2024-10-12:
Strange, I guess it will be fixed by clang team at some point.
You can also switch to GCC, options are compatible.

I did my testing on this arch, using a (`Raspberry Pi 4 Model B Rev 1.2`):
```
$ clang --version
clang version 7.0.1-8+rpi3+deb10u2 (tags/RELEASE_701/final)
Target: armv6k-unknown-linux-gnueabihf
```

closed 2024-10-12 by Guenael