Guenael/rtlsdr-wsprd #142: Cannot run on Pizero W - update Makefile

Claudio-Sjo, 2024-12-09:
Change compilation flag in Makefile for RPi zero
`CC = clang
CFLAGS = -O3 -std=gnu17
LIBS = -lusb-1.0 -lrtlsdr -lpthread -lfftw3f -lcurl -lm

ifeq ($(findstring armv6,$(shell uname -m)),armv6)
# Broadcom BCM2835 SoC with 700 MHz 32-bit ARM 1176JZF-S (ARMv6 arch)                                                                                       PI_VERSION = -DRPI1 --target=arm-linux-gnueabihf -mcpu=arm1176jzf-s -mfloat-abi=hard
else
# Broadcom BCM2836 SoC with 900 MHz 32-bit quad-core ARM Cortex-A7  (ARMv7 arch)
# Broadcom BCM2837 SoC with 1.2 GHz 64-bit quad-core ARM Cortex-A53 (ARMv8 arch)
PI_VERSION = -DRPI23
endif

# Note
#   gcc is a bit faster that clang on this app
#   for dbg: -Wall -fsanitize=address

OBJS = rtlsdr_wsprd.o wsprd/wsprd.o wsprd/wsprsim_utils.o wsprd/wsprd_utils.o wsprd/tab.o wsprd/fano.o wsprd/nhash.o

TARGETS = rtlsdr_wsprd

.PHONY: all clean

all: $(TARGETS)

%.o: %.c
        ${CC} ${CFLAGS} $(PI_VERSION) -c $< -o $@

rtlsdr_wsprd: $(OBJS)
        $(CC) -o $@ $^ $(LIBS)

clean:
        rm -f *.o wsprd/*.o $(TARGETS) fftw_wisdom.dat hashtable.txt selftest.iq

install:
        install rtlsdr_wsprd /usr/local/bin/rtlsdr_wsprd`

Guenael, 2024-12-10:
Could you please provide a Makefile that works also for other architectures? (you're focusing only on Arm devices with this Makefile)

Guenael, 2024-12-10:
I made this update, please let me know if it works for you.
https://github.com/Guenael/rtlsdr-wsprd/blob/main/Makefile#L9-L22

Claudio-Sjo, 2024-12-10:
It doesn't work.
The problem is clang compiling for the wrong architecture, you need to specify
--target=arm-linux-gnueabihf -mcpu=arm1176jzf-s -mfloat-abi=hard

73 de SA0PRF

Guenael, 2024-12-10:
Update completed : https://github.com/Guenael/rtlsdr-wsprd/blob/main/Makefile#L12

closed 2024-12-12 by Claudio-Sjo